opentx/opentx #9015: FrSky Archer Access Plus Receivers

AlphaCharlyX, 2023-06-01:
OTA Updates and connecting to the option menue doesn't work with the new FrSky Archer Access Plus Receiver.

Is it possible to fix this?

BlueBlue58, 2023-06-26:
You have to power on in a certain order for it to work. Connect radio and receiver normally. disconnect receiver then go to update and start OTA, then renonnect power to receiver and it should find the OTA. Your esc will beep continuously until update successful. Hope this hepd?

AlphaCharlyX, 2023-07-02:
No, sorry it does not work either.

remcat07, 2023-09-03:
I see the same issue , OTA Updates and connecting to the option menue doesn't work with the new FrSky Archer Access Plus Receiver. (OTX 2.3.15; X10S Express,  Archer Plus GR8 )

rieden1, 2023-09-04:
Same issue here on R6 Archer-Plus (reported and discussed on RCG from here: https://www.rcgroups.com/forums/showthread.php?3639447-FrSky-Archer-and-Archer-Plus-Series-ACCESS-Receivers/page168#post51445857) - no Rx Options menu, no OTA function (not even listed in the options when I select the firmware file in the system menu). Rx also does not report a firmware version in the system menu data screen.

I reflashed the firmware with latest version via the S-Port which worked, but didn't resolve the problem. I also reflashed the ISRM firmware because I saw a new version came out a few days ago on the Frsky website. Again, the flashing was successful but it didn't resolve the problem.

Config:
Horus X10sSE
OTx 2.3.14
ISRM firmware ISRM-S-X10E_2.1.0 and ISRM-S-X10E_2.1.9
Rx Firmware Archer_Plus_R6_GR6_R6FB_GR6FB_1.0.5

AlphaCharlyX, 2023-09-04:
Does EdgeTX works?

ToniCupra, 2023-09-04:
> Does EdgeTX works?

Not perfectly.  I can switch the gyro on and off via lua but no calibration or so

raphaelcoeffic, 2023-09-04:
> FrSky Archer Access Plus Receivers are not fully supported

The truth is: FrSky does not support neither OpenTX nor EdgeTX with any useful information. If anything has changed prior to what used to work, it's basically toast.

One of 2 things would need to happen:
- FrSky starts disclosing information allowing open source devs to support these new things,
- *Someone* takes care of maintenance and reverse engineers these changes.

Until then, please consider FrSky specific hardware / radio tech as **not supported**.

rieden1, 2023-09-06:
Mike Blandford has done a superb investigation and tracked down the problem - passing on his comments here (with his permission):
_The "Waiting for Rx" problem is as follows:
openTx first requests the Rx hardware information. This includes the receiver type as a numeric value. openTx then looks up the name of the receiver in a table. If the numeric value is beyond the end of the table, then openTx abandons reading the actual receiver options.
Currently, the Archer plus receivers are not in the openTx table. The table is currently:_ 
  "X8R",
  "RX8R",
  "RX8R-PRO",
  "RX6R",
  "RX4R",
  "G-RX8",
  "G-RX6",
  "X6R",
  "X4R",
  "X4R-SB",
  "XSR",
  "XSR-M",
  "RXSR",
  "S6R",
  "S8R",
  "XM",
  "XM+",
  "XMR",
  "R9",
  "R9-SLIM",
  "R9-SLIM+",
  "R9-MINI",
  "R9-MM",
  "R9-STAB", // R9-STAB has OTA
  "R9-MINI-OTA", // this one has OTA (different bootloader)
  "R9-MM-OTA", // this one has OTA (different bootloader)
  "R9-SLIM+-OTA", // this one has OTA (different bootloader)
  "Archer-X", // this one has OTA (internal module)
  "R9MX", // this one has OTA
  "R9SX", // this one has OTA

_I've gone through a disassembly of Ethos and worked out the table should be:_

"---",
  "X8R",
  "RX8R",
  "RX8R-PRO",
  "RX6R",
  "RX4R",
  "G-RX8",
  "G-RX6",
  "X6R",
  "X4R",
  "X4R-SB",
  "XSR",
  "XSR-M",
  "RXSR",
  "S6R",
  "S8R",
  "XM",
  "XM+",
  "XMR",
  "R9",
  "R9-SLIM",
  "R9-SLIM+",
  "R9-MINI",
  "R9-MM",
  "R9-STAB", // R9-STAB has OTA
  "R9-MINI-OTA", // this one has OTA (different bootloader)
  "R9-MM-OTA", // this one has OTA (different bootloader)
  "R9-SLIM+-OTA", // this one has OTA (different bootloader)
  "Archer-X", // this one has OTA (internal module)
  "R9MX", // this one has OTA
  "R9SX", // this one has OTA
  "TDMX",
  "TDR18",
  "TDR10",
  "TDR6",
  "TDR12",
  "TDSR12",
  "TDSR18",
  "---",
  "---",
  "---",
  "---",
  "---",
  "---",
  "---",
  "---",
  "---",
  "---",
  "---",
  "TWM8",
  "TWSR12",
  "TWR12",
  "TWGR6",
  "R12",
  "TWR6",
  "TWR8",
  "---",
  "---",
  "---",
  "---",
  "---",
  "---",
  "---",
  "---",
  "SR10 Plus",
  "R10 Plus",
  "GR8 Plus",
  "R8 Plus",
  "SR8 Plus",
  "GR6 Plus",
  "R6 Plus",
  "R6M (ESC DC)",
  "Rs PlusRS Mini",
  "R6FB"

_Not sure about the last 2 entries!

MIke_

If I understand this correctly it just needs an OTx build with this updated table to fix the problem. I'm guessing this applies equally to ETx, but that's just my guess.

Could someone pick this up?

Peter

raphaelcoeffic, 2023-09-07:
LoL, I went through the same thing (disassembling Ethos) a couple of months ago for that same table. Nice to hear that is that's needed.

3djc, 2023-09-07:
Since Otx does have an FrSky employee leading it, it's really up to FrSky if they want that supported in Otx or not, no need for disassembly or anything like it

rieden1, 2023-09-07:
Are you saying that no one will pick this up even though the required information has been provided?

3djc, 2023-09-07:
As said above, if you want it in Otx, you should convince FrSky

rieden1, 2023-09-07:
I didn't realise that the OTx development was so firmly under the control of FrSky - I had always believed that OTx responsed to the wants of its users. A lot of users would like this change. We have provided you with the required information to make the change, but you have chosen to sulk instead. Do you speak for all of the OTx team or is this just a personal hate campaign?

kilrah, 2023-09-07:
The main developer behind OpenTX, who is actually capable of developing most major things and keep the project maintained has been employed by FrSky, who initially was interested in keeping OpenTX maintained, but then moved to wanting him to pretty much exclusively work on their own Ethos and leaving no time for OpenTX. OpenTX has been mostly in limbo since then since there's no one who has joined the project to take that work, those who could have decided to fork it into EdgeTX instead, which makes sense since having no time to work on OpenTX also meant no time to review and integrate 3rd-party contributions.

rieden1, 2023-09-07:
Thanks for that Kilrah. I'm led to believe that this is a trivial change - just replace one list with another list and run the build. Is there really no one who can do this any more? Would the change be more likely to happen in ETx? To date I've not moved to ETx because my radio (Horus X10sSE) isn't a touch-screen one and I found the UI less convenient, but this might finally be a reason for migrating!

3djc, 2023-09-07:
For merges to occur, and even more releases, it needs works from project owner, and that won't happen as long as FrSky doesn't allocate some of his time for that. That's merely facts, not the childish vendetta you claim about even cross posting into forums...

Changes are more likely to happen in Edgetx, if you find a volunteer to do it

bsongis, 2023-09-22:
Changes done. Sorry for having so little time.
There was a stupid sanity check which prevented further receivers to be compatible. 
I will release 2.3.16 soon

closed 2023-09-22 by bsongis

rieden1, 2023-09-22:
Bertrand - big, big thanks for this and apologies for the ranting! Will 2.3.16 also include support for the angular position data from gyro receivers, or is that something Frsky want to keep as an "Ethos-only" thing?

Thanks again,

Pete

3djc, 2023-09-22:
Pitch and roll will be in 2.3.16

rieden1, 2023-09-22:
And yaw?

Chaps, again many apologies - I've done some pretty childish ranting on this issue!

Pete

3djc, 2023-09-22:
Currently not, will see if available

3djc, 2023-09-22:
Apparently yaw sensor is not avalable, I'm told not in Ethos either

bsongis, 2023-09-22:
No Yaw, I can confirm

rieden1, 2023-09-22:
OK, thanks for checking. And again, apologies for my previous ranting.

Steven-AZ, 2023-11-12:
Any chance of releasing 2.3.16 any time soon?  Or will that be done/announced somewhere else?

QuadStryke, 2024-04-09:
Still no 2.3.16? It's been more than half a year the last time we heard something sadly

Steven-AZ, 2024-04-09:
I don't think it will ever come. Time to face it: OpenTX is dead.

FrSky clearly does not care, or they would have asked Bertrand to do this very minor update a long time ago. They probably assume that any OTX problems will result in an accelerated switch to Ethos (and they may be right).

I was very fortunate to have Mike Blandford create a custom version (2.3.15b) for my QX7 that allows me to use Archer Plus receivers. However, I have this nagging concern that there might be other issues (beyond the Rx table) lurking in the background that may cause problems. I had my first radio-caused crash last week (spontaneous failsafe condition in flight) using an Archer Plus R8. Coincidence? Hopefully.

I will probably get an Ethos-based Tx eventually, but at this point I have soured on FrSky as a company.

AlphaCharlyX, 2024-04-09:
Or switch to EdgeTX

geofrancis, 2024-04-09:
if you want any new features you need to change over to edge tx, its basically the same, all your models will transfer over without issue.

Steven-AZ, 2024-04-09:
Thanks for the tip. Indeed, that would be an option. So far I've hesitated to try it, as I saw it as a temporary switch before ultimately going to Ethos. But if it's that easy (just installing it on my QX7?) I will consider it.

What about erskyTx vs EdgeTx: issues or advantages of either?

geofrancis, 2024-04-09:
I dont know if your models will copy across to er9x.

bigtwin, 2024-04-09:
I dont know about _erskyTx_, but I switched just lately from OTX to ETX. Actually you can treat it being the same. ETHOS is definetly way different. Though, ETX 2.9 does not feel that smoothly when navigating thru the menus and their items. It might be different (better) if you have a touch display.

rieden1, 2024-04-10:
> if you want any new features you need to change over to edge tx, its basically the same, all your models will transfer over without issue.

It's my understanding that ETx doesn't fully support Advance Plus receivers either - it has exactly the same issue.

I've been using Mike Blandford's modified OTx 2.1.15 in my X10s EX for nearly a year with no issues whatsoever (I think it was me that first asked him to do it). Big shout-out for Mike B - many, many thanks!

geofrancis, 2024-04-10:
> It's my understanding that ETx doesn't fully support Advance Plus receivers either - it has exactly the same issue.

https://www.rcgroups.com/forums/showpost.php?p=50736175&postcount=4469

QuadStryke, 2024-04-10:
On EdgeTX Discord they say support was added with version 2.9.2

QuadStryke, 2024-04-10:
Bertrand just answered me via email:

Right, the problem is the lack of time, I am full time on Ethos which is growing fast!
But promised, I will release 2.3.16 when things calm down on Ethos 1.5

bigtwin, 2024-04-11:
It was OK to publish his mail?

QuadStryke, 2024-04-11:
Well I don't think he would write something confidential via email to a random guy. And it's public interest if we will ever get a 2.3.16

pr-newman, 2024-04-23:
> Changes done. Sorry for having so little time. There was a stupid sanity check which prevented further receivers to be compatible. I will release 2.3.16 soon

When is soon, please?  You closed this as if the fix was already out there.  Can you also re-open it until the official 2.3.16 is released, or state clearly that it will never be released?

QuadStryke, 2024-04-23:
@pr-newman it's not Bongis, it's bsongis...

pr-newman, 2024-04-23:
> @pr-newman it's bsongis, not Bongis...

Thank you! 😊

BompaBert, 2024-10-30:
Am looking for Mike's compiled 2.3.15 versions however Openrcforums seems down. Any chance they still available for download somewhere else?

rieden1, 2024-10-30:
That would depend what Tx and other options you wanted. I have a Horus X10sSE build if that helps. But alternatively contact Mike on RC Groups - you'll find he contributes a lot to the OpenTx threads.

BompaBert, 2024-10-31:
Yup, same here, Horus X10 Express. Getting into the RX Options menu would help a lot. Roll and pitch would be a nice have too.

BompaBert, 2024-11-02:
Well, Bertrand has merged a pull request on Mike's 2.3.15B about a week ago. Till further updates, am running Companion 2.3.16N7 (for some reason, the server doesn't let me download version N13) and for the radio, the X10express 2.3.16N13.bin file.
It gives me access to the + receiver 'options' menu and gives me 'pitch' as well.

BompaBert, 2024-11-04:
An update on this. Version 2.3.16N13 gives me problems discovering telemetry. If everything is already discovered before the upgrade, all is ok but if you delete and rediscover using this version, multiple of the same sensors -3 voltage sensors in my case - or not seen. The discovery also takes much more time. because of this, am using now Mike's 2.13.15B. Works like a champ and besides the RX + options menu, roll + pitch are there too  -THANK YOU MIKE!