opentx/opentx #9016: Receiving "Tx antenna problem!" message on X9D+ transmitter.

rscarawa, 2023-06-02:
I have an X9D+ from around 2015 that is throwing an error "Tx antenna problem!" every 10 seconds.  I just did an upgrade from 2.0.17 to the 2.3.5 release.  It seems the error is not a real error and it is linked to the sw code, possibly in /telemetry/telemetry.cpp.  I understand that some X9D+ do not have the proper HW to determine if the antenna is indeed bad.  Is this the case?  If so, what would be the workaround on the 2.3.15 release?  Does the code need modification for these radios with different hardware?  Would installing the X9D firmware instead of the X9D+ solve this problem?  I did a downgrade to 2.0.17 and the error went away but I think the code did not check for antenna health at that time.

**Describe the bug**
Tx antenna problem!" every 10 seconds after upgrade from 2.0.17 to 2.3.15.

**To Reproduce**
Steps to reproduce the behavior:
1. See above.

**Expected behavior**
No reported error message

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version 2.3.15
 - Hardware Frsky X9D+ 
 - OpenTX Companion 2.3
[image]

jtaylor2, 2023-06-02:
In Companion go to settings and under Radio Type Build Options select noras.  Download firmware again and flash.  Do NOT install X9D or any other firmware except that which matches your Radio.

Jim

rscarawa, 2023-06-03:
Resolved.  Thank you so very much for the assistance.

closed 2023-06-03 by rscarawa