opentx/opentx #9019: companion update failed win 10

normtx16s, 2023-07-24:
The issue tracker here is for devlopment of OpenTX. This is not a
support site, only a bug tracking system. Please use support groups
like [rcgroups](https://rcgroups.com/) or
[openrcforum](https://www.openrcforums.com/forum/). See also our
[manual](https://opentx.gitbooks.io/manual-for-opentx-2-2/).

For quick questions you can also join our general chat.

Issues that fall into the category of asking for help, will be
generally closed as invalid without an answer.

JFingerle, 2023-09-06:
Please close this nonsense issue. Thank you!

normtx16s, 2023-09-06:
An issue is only nonsense to someone who knows all the answers!
Thanks for all your help...
...

Closed #9019<https://github.com/opentx/opentx/issues/9019> as completed.

JFingerle, 2023-09-06:
@normtx16s In this case the issue description is a placeholder text, thus the issue is nonsense to me. In case your original text was edited by someone else I apologize. Have a good one!