opentx/opentx #9021: OpenTX & 2.19 Branch not downloading firmware for Turnigy 9xr (Not Pro)

MW0SBX, 2023-08-24:
OpenTX & 2.19 Branch not downloading firmware for Turnigy 9xr (Not Pro)

Firmware opentx-9xr128-en does not seem to have ever been downloaded.
Release 2.1.9 (2016-09-14) is available.
Do you want to download it now?

You say yes and it fails.

Download failed: Error downloading http://downloads-21.open-tx.org/firmware/getfw.php?fw=opentx-9xr128-en.hex - server replied: Bad Gateway.

vladkvit, 2023-09-04:
Same issue here. Sorry in advance if this doesn't add to the discussion, but I'd also love to know how to download the older (2.1.x) firmware for my atmega-based 9xr.

ulysse31, 2024-09-26:
Hello,

All firwmwares for version 2.1 seems to have been deleted from the download servers of open-tx ...
I suppose the firmware should be located in : https://downloads.open-tx.org/2.1/firmware/
But nothing ...
One think is to stop  support ... deleting all builds and turning them impossible to download is another.