opentx/opentx #9022: Clone option for MPM DSM Protocol

johnnym007, 2023-08-25:
Add an additional option to the DSM MPM screen to indicate whether it should operate in Normal or Cloned mode (similar to the FRSKY cloning). TX should then send the value in the optionValue (byte3) bit5 - for DSM currently:

Bits 0..3 num_ch
Bit 4 Not Used
Bit 5 Not Used
Bit 6 Frame Rate
Bit 7 Max Throw

Additionally it would be good if the bind button was disabled/hidden when Cloned mode is selected to avoid any uncertainty of whether a bind operation is actually required.
Initial suggestion is to add something like (with the values Cloned[1] & Normal[0]):
[image]

User would need to:

1. Clone the TX using the DSM_RX protocol using the RX_num to identify the model match number.
2. Go to the DSM protocol select 'Cloned' and then match the subprotocol and RX_num to the model being cloned.
Note: Spektum XOR the model match number with the last byte of the ID, hence the need to select the correct model match number when cloning the TX (so that the ID can be determined without model match).

I have already added the ability to Clone and Erase as DSM TX in my own fork of the MPM repo. If it would be possible to have a binary with this feature added to allow me to test that would be great (X9d+ & TX16s).

https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/396

3djc, 2023-08-25:
1) Won't be adding that to Otx, but to EdgeTX

2) Thats the other way around, I need to working MPM build to do the UI

johnnym007, 2023-08-25:
> Thats the other way around, I need to working MPM build to do the UI
Ok - I'll add the the necessary code to read the new option value.  My assumption is that the 'Cloned' 'Normal' setting would always be present on the TX (whether it was implemented in the MPM firmware or not).  If we want this setting to only be visible if it is supported by the MPM firmware (and a TX has already been cloned) then I will need to add this to the flags byte.

How do you want the MPM build? I can give you a link to my fork or provide a binary?

3djc, 2023-08-25:
If not implemented in MPM firmware, then you cannot display cloned (the UI will try to change that bit, but MPM will ignore it and resend it blank next telem frame, so the UI will display normal)

Why does Bind need to be hidden ?

johnnym007, 2023-08-25:
> Why does Bind need to be hidden ?

Just thought it would be a nice feature, but I guess I can disable it from doing anything from the MPM.

3djc, 2023-08-25:
I mean why does it need to be hidden ? You cannot bind cloned ?

johnnym007, 2023-08-25:
Good question.  I guess you could bind in cloned mode and that would work between cloned MPM modules.  I'm not sure what a Spektrum TX would do here - I'll need to try.

pascallanger, 2023-08-25:
You can use the invalid sub_protocol flag if an ID has not been sent. When this is the case just use this line : "SUB_PROTO_INVALID;" The radio will display the message.

johnnym007, 2023-08-25:
Thanks Pascal - Sorry for all the questions.

3djc, 2023-08-26:
You should be able to try with that:
[x9d+ dsm cloned.zip](https://github.com/opentx/opentx/files/12444927/x9d%2B.dsm.cloned.zip)

johnnym007, 2023-08-26:
Wow - thanks @3djc.  I'll give it a try later today.

johnnym007, 2023-08-26:
Just tested it very quickly.  I have two receivers connected one bound with the normal ID and one with the cloned ID and I can switch between the two with the new Subtype control. :-)  Thanks again @3djc & @pascallanger really appreciated!

I need to do some additional testing to ensure i've not inadvertenlty broken anything...but everything looks good so far.  Also need to test the DSM2 subprotocol and have located my old DX7s TX for this purpose.

The only limitiations I can see so far are:
1. OrangeRX where we obviously cannot manipulate the IDs when cloning:
```
//Fix for OrangeRX using wrong DSM_pncodes by preventing access to "Col 8"
if(sop_col==0 && sub_protocol != DSMR)
{
cyrfmfg_id[rx_tx_addr[0]%3]^=0x01;					//Change a bit so sop_col will be different from 0
sop_col = (cyrfmfg_id[0] + cyrfmfg_id[1] + cyrfmfg_id[2] + 2) & 0x07;
}
```
2.  Maximum of 63 model match numbers (probably not an issue for most)

johnnym007, 2023-08-26:
@3djc - I think Cloned & Normal should be switched, so that the default option when the protocol is selected is Normal (0).

3djc, 2023-08-26:
Easy, will do

3djc, 2023-08-27:
Here is a TX16S version, beware it is 2.10 based, so you 'll wxant to try on a copy of your SD

[TX12S-DSM-CLONE.zip](https://github.com/opentx/opentx/files/12447590/TX12S-DSM-CLONE.zip)

gagarinlg, 2023-08-27:
Does is makes sense to have this lengthy discussion of a, very nice, feature for EdgeTX in an OpenTX Github issue?

3djc, 2023-08-27:
It started here, but is implemented in Edge. Sure, we could have done it another way, but what matters is it done now, no ?