opentx/opentx #9025: Taranis Radio not working as USB Joystick for computer simulator software anymore

Tigerlie, 2023-09-13:
since the latest OpenTX 2.3.15 on the Taranis X9D Plus radio, it will no longer work as a Joystick to practice on computer simulators.

everything worked before on version 2.2.4.
everything else is the same, computer, simulator, USB cable etc.

the radio asks how i want to connect USB, i choose human interface device, but nothing works on the computer.

3djc, 2023-09-14:
OpenTX has now it's own USB id, my guess is you need to reinstall the driver with the new id

Tigerlie, 2023-09-16:
i appreciate your insight,
i was not able to get it to work, maybe someone has another idea.
also it seems like there is no way to get it to work at all, on a MacBook.
why did the developers change this, it was working nicely before, i think there should be a final update for the FrSky Taranis radios, while Edge is great for RadioMaster radios.

kilrah, 2023-09-17:
It was required because the old method also had issues for some. You should be able to find guides on how to go from one driver to the other from back when that happened, although that was a few year ago already.

Tigerlie, 2023-09-17:
i see, but this seems like there is no possible way to change it for some retro mac computers with around2012ish MountainLion or Mavericks OS.

Tigerlie, 2023-09-25:
i'm happy to report, the Taranis Radio is now working in USB Joystick mode. (for whatever reason)
i was tinkering on the weekend and found that the radio is still recognized as a Joystick, then i tried again with the simulators, they recognized a new never seen before joystick, i calibrated it as per simulator procedure, and it's working great now. it seems like there is no issue with the firmware after all. things happen.

closed 2023-09-25 by Tigerlie