opentx/opentx #9029: companion 2.1.9 link not working

NickKlotz1, 2023-12-11:
hello, im a new user to opentx and id like to flash my turnigy 9x with opentx, but im finding that companion 2.1.9 is not downloadable on the opentx website. the windows installer link does not work. is there another way to get companion 2.1.9?

pfeerick, 2023-12-20:
Looks like the link is wrong on the website. The proper link is https://downloads.open-tx.org/2.1/companion/companion-windows-2.1.9.exe

DAMskill, 2024-04-03:
After spending some days and almost giving up.
Used eePe EEPROM Manager to backup the Turgigy TGY 9X
I finaly stumbled upon your link. 

A Working version of the ooh so precious OpenTX  Companion2.1.9  
Time to give this old beast new life
Thanks a lot

ulysse31, 2024-09-25:
Hello,

I know that the old 2.1.9 is now "unsupported" but right now the companion cannot even download firmwares ("Unable to check for updates" error) ...
Is there some way to download firmware manually somewhere ? I'm searching for opentx-9x-en 2.1.9 ... did not find it anywhere on the internet ...
Thanks a lot for the help,

--
Ulysse31

willyfpv, 2024-12-06:
Hi! im in the same situation did you find anything ? I have a tgy9x with stock board

UmbraAtrox, 2024-12-08:
Is it because https://www.open-tx.org/2021/10/01/companion-downloads 

> Since Sept 30th, 2021 OpenTX Companion for Windows is not able to check for updates / download them or download firmware anymore.

> This is caused by the combination of [the expiration of a Let’s Encrypt root certificate](https://letsencrypt.org/docs/dst-root-ca-x3-expiration-september-2021/) and the fact that the OpenSSL library that has been bundled with the Windows version of Companion is too old to support the transition method that Let’s Encrypt has used. Companion thus can’t reach the download server anymore.
We have rebuilt Companion 2.3.14 with a newer version of the libraries to fix the problem.

and we'd need a 2.1.9 with the updated libraries?

willyfpv, 2024-12-08:
Uh ok 
I was trying to update the turnigy firmware with some functions that previously I've not enabled that's all.
If the library is not accessible anymore I think I cannot do it,it is what it is 
Thanks for the reply

ulysse31, 2024-12-08:
Hello,

Why not just put available the latest binaries for all the radios ? It does not cost any extra ressource, no special maintenance needed, just put the binaries available for download, people can then flash there old radios manually.
The fact that the companion app would be mandatory does not make sense to me, could you explain ?
Thanks a lot

3djc, 2024-12-08:
Firmware were compiled on demand (over 10k different variants had been asked) During last server move, the 2.1 compilation environment has not been reinstalled, mostly because this much smaller server isn't quite up to par to do it). So unless users have a backup of their firmware, those cannot be generated from the server anymore

ulysse31, 2024-12-08:
BTW: It's a PITA, but I had to download the sources, install the toolchains, and compile it myself to get the binary for my RC ... a bit sad to have to do all this, when it would cost you zero to simply have the a repo tree (on github or elsewhere) with the latest version. Not supporting a version anymore only means no more bug patch or feature enhancements ... it does not mean making all binaries disapear ... ^^"

willyfpv, 2024-12-08:
> Hello,
>
> Why not just put available the latest binaries for all the radios ? It does not cost any extra ressource, no special maintenance needed, just put the binaries available for download, people [...]

That's what I was thinking

willyfpv, 2024-12-08:
> BTW: It's a PITA, but I had to download the sources, install the toolchains, and compile it myself to get the binary for my RC ... a bit sad to have to do all this, when it would cost you zero to [...]

Your compiled firmware was for a turnigy 9X?

ulysse31, 2024-12-08:
10k variations ? ... hmmm ... strange since I did not see 10k options on the build source readme ... ^^" maybe some extra extensions ? anyways ... please look at other projects like openwrt ... they do not use or need a "companion" to build firmware ...
I do see the advantage of it for "non-techie" people ... but downloading binaries for routers/embedded devices ... is the basics ... to me at least ...

gagarinlg, 2024-12-08:
> BTW: It's a PITA, but I had to download the sources, install the toolchains, and compile it myself to get the binary for my RC ... a bit sad to have to do all this, when it would cost you zero to [...]

You forget one thing, it costs time and effort. Nobody is getting any money for it, but people spend their free time on these kinds of projects.  Sadly a lot less people use OpenTX or EdgeTX than openwrt, so we have a lot less people helping with the work.

You can ask for something but you are not entitled to get anything.

ulysse31, 2024-12-08:
> Your compiled firmware was for a turnigy 9X?

Yes ^^'
I followed the instructions available here : https://github.com/opentx/opentx/wiki/OpenTX-2.1-Linux-Build-Instructions
I had to create a chroot with the correct debian version (using debootstrap) in order to get all the correct package versions, so yes, *it was clearly a big PITA*
But that was in September, since then I played a bit with my old t9x, gave it to a cousin and now bought an jumper T15 ^^" (and I'm using edgeTX now ...)

ulysse31, 2024-12-08:
> You forget one thing, it costs time and effort. Nobody is getting any money for it, but people spend their free time on these kinds of projects. Sadly a lot less people use OpenTX or EdgeTX than [...]

I totally understand that opensource is time and effort consuming (I know it very well ^^') but that's a one time job, that can be (and generally is) automated via scripts.
Like a wise green little old humanoid life said one day "do or do not, there is no try".

kilrah, 2024-12-08:
> 10k variations ? ... hmmm ... strange since I did not see 10k options on the build source readme 

See the number of options in radio settings, number of languages... you can calculate the number of possible combinations yourself.

> but that's a one time job, that can be (and generally is) automated via scripts.

It was scripted of course but environments change, dependencies break, infrastructure changes etc so some maintenance is needed.
OpenTX stopped working on anything 9x/AVR 8 years ago back in 2016.

ulysse31, 2024-12-08:
> It was scripted of course but environments change, dependencies break, infrastructure changes etc so some maintenance is needed. OpenTX stopped working on anything 9x/AVR 8 years ago back in 2016.

I think you may mislead people saying that : companionTX did stopped working ... OpenTX (the firmware on the RC) are still working as of today ! and *technically* anyone can continue to flash 2.1.9 firmware on old 9x/AVR => **it will still work !**

Anyways, to me no more need for this ^^

kilrah, 2024-12-08:
Sure if you can build and flash yourself it'll work just fine since neither the code or device has changed. My point is the infrastructure required to build/serve is long deprecated and now unavailable.