opentx/opentx #9033: OpenTX 2.3.15 EU-Firmware for Horus X10S Express is not available

FredSey, 2024-01-20:
I've tried to flash OpenTX to my Horus X10S Express.
On the Radio Profile for X10S Express I can't choose the EU-Version (LBT).
So I have changed to the X10/X10S Radio Profile.
After flashing I have actually an Error "wrong PCB detected" (falsche PCB erkannt)
The SD-card content seems to be okay (480x272-2.3V0039)
What can I do?

**General information**
 - OpenTX version 2.3.15
 - Hardware Horus X10S Express
 - OpenTX Companion 2.3.15

3djc, 2024-01-20:
Since all EU option does is prevent d8 to appear, and x10 express does NOT offer d8, it is all vey logical.  Otx EU option has nothing to do with lbt. Lbt comes with eu version of module firmware

closed 2024-01-20 by 3djc

FredSey, 2024-01-20:
So I have to choose the Horus X10S Express Radio Profile an flash again.
After this I have to flash the internal  ISRM-S-X10 Modul to LBT.
I this correct?

3djc, 2024-01-20:
> So I have to choose the Horus X10S Express Radio Profile an flash again. After this I have to flash the internal ISRM-S-X10 Modul to LBT. I this correct?

Absolutely

FredSey, 2024-01-20:
Thank you for Information an quick help.
I try this now an I hope it will fix the problem.