opentx/opentx #9035: Is OpenTx dead?

stevefoxuk, 2024-02-05:
People have abandoned OpenTx in droves for Edge Tx but some like myself have waited with patience for 2.4 which has been literally years now.

Is it ever coming or is OpenTx now dead and we should all jump ship to get the features others have had for a very long time in Edge Tx?

3djc, 2024-02-05:
I think it is safe to say that 2.4 is very unlikely to come out.
OpenTX owner now works for FrSky, so basically changes will only happen when FrSky sees fit.

phamiltonsmith, 2024-02-05:
And if OpenTx is dead, what effect will that have on our Frsky Radios? Will the become obsolete?  As I fear they will...

I for one just migrated over to Edge... I tried to hold on as long as I could, but no longer

MlCHEL1234, 2024-02-05:
And if OpenTx is "only" about troubleshooting etc., what impact will this have on our Frsky radios? Will they become obsolete?  I don't think they will...

At some point, every product that is subject to permanent further development virtually in real time will have reached a level of maturity that is very difficult to improve. The proprietary systems of the market leaders have never reached this state, there has never been enough further development. Edge has not reached this level of maturity either. There are still years to go. The only improvement that OpenTx does not have is touchscreen support. But since I mainly use my transmitter for flying and not for permanent programming, Edge offers me nothing that I have to have. 

There is still a large majority in our club who still have both feet firmly on the ground and smile pityingly at the few nerds...

bigtwin, 2024-02-06:
I dont agree here. There are quite some more little tiny improvements oustanding in OTX. And as even those smaller and most probably easy to fix issues were not considered in OTX, I switched to ETX some month ago. Just because I knew those things have already been fixed/implemented in ETX. A touchscreen is not of interest for me and I never ever would count me to those "few nerds".
However, its clear that some manufactors stopped developing their Software with way more problems und tons of missing features. Eventhough the development of OTX could have a more graceful end, I am still thankfull for its influence to the market of transmitters. For me ETX is the logical continuance of OTX.

3djc, 2024-02-06:
Very clearly, and in total compliance with GPL, and after discussing with Otx team, EdgeTX started from Otx code, so everything that was in Otx, is in EdgeTX.

If you talk only BW radio, there is a ton of nice features that have been added, but really an Otx user will feel at home on an EdgeTX radio.

There as been more profond UI changes for colorlcd radio, but those where needed to support touch radios, so I can understand that some Horus users feel EdgeTX doesn't quite match Otx there, but said simply, it is not Otx or EdgeTX team doing, this is simply FrSky deciding that  Otx is a competitor to Ethos and therefore needs to be stopped, and (more than) clearly not supporting EdgeTX.

Very little point complaining here or on EdgeTX side. (but again, I can understand some frustration, you need to take it to FrSky)

MicheleVilla, 2024-02-06:
> This is a very good point. I think it was OTX which made FrSky great. Not only the Software but as well the very passioned community. Next time I am purchasing new hardware, I will check with Radiomaster and I hope they continue the good releationship with the community. Currently I would rather switch system than buying new from FrSky.

I agree
FRSKy started out producing modules compatible with other brands
OpenTx has allowed it to produce large profits by selling many radios with an innovative and economical system
But FRSKY is a company that produces profits and serves its own interests
When they realized that they in turn were being copied, he closed everything and developed his system taking advantage of the OpenTx experience (whoever makes it waits for it)
I and the community that contributed to OpenTx and purchased Frsky now feel a little betrayed and used.
Now I have switched to radiomaster, but I still have a lot of Frsky hardware which from a telemetry point of view is still excellent compared to new solutions, but above all it constitutes an investment that I made believing in that company which has now somehow taken a completely different direction
EdgeTx is the natural continuation of openTx, let's support it and hope for a bright future.

phamiltonsmith, 2024-02-06:
I have enjoyed using FRSY/OpenTx products for years.  Overall, I can say I have been a happy customer...

Until now: a week ago I purchased an Archer Plus GR6 receiver - I now find it is not supported by the current FW. To use the receiver, I am being asked (by Horus support) to run a nightly version...  So, FRSKY **appears** to be selling products (at least in my case) incompatible with current FW versions. Sounds like a pretty honest practice to me...

It is unclear when the next FW version will be out, so it looks as though I just donated $50 to FRSKY...

My jump to Radiomaster will be soon... 

I know this is a Forum for OTX ISSUES but I am angry and after days of flashing and re-flashing I am nowhere closer to getting my receiver to work...

mpjf01, 2024-02-07:
I have several of these receivers that I use with Opentx. What is the
"current firmware" that you refer to?

pfeerick, 2024-02-07:
Most likely a nightly firmware downloaded using a nightly version of Companion from https://github.com/opentx/opentx/releases/tag/nightly

It would be needed in order to have https://github.com/opentx/opentx/commit/29dd63af8bf7217033e1b5a553db0ccb9454546a and https://github.com/opentx/opentx/commit/6fef3b6849250bc3185553381b8a94f63da5fa4f, which allows for OTA firmware flashing and ACCESS configuration options for newer Archer receivers.

If you own a Archer Plus not all is lost... can use any transmitter that has a internal MPM, a transmitter with a module bay and external MPM, and use it in ACCST D16 mode. You won't get *all* of the features that ACCESS will give you, but at least you'll have a working receiver.

stevefoxuk, 2024-02-07:
It looks like my patience and loyalty to OpenTx was a waste of time then, looks like another will jump ship and install Edge Tx to get the features promised as coming soon for the last 4 years in OpenTx. 

OTX has gone from the king of Tx software so nothing so fast and its the developers who are responsible.

bigtwin, 2024-02-07:
Dont judge that hard! It's all privat and personal engagement.   Have you ever donated for OTX? ;-)

In my mind the only questionable thing is the end itself. It could have been more professional and more respectable. Leave evertyhing as it is, communicate via timeout and simply let OTX fade out, has surely room for improvement. 

But as stated before - ETX seems to be next evolution OTX. The logical successor. I am very happy, that even with ETX2.9 my topics are already covered in comparison to OTX2.3. Looking forward to discover more from ETX! It is more vital than OTX ever was at least during the time I watched it.

MicheleVilla, 2024-02-07:
Unfortunately, even in other software it has happened that private companies have benefited from the contribution of a community and then suddenly closed down and continued on a different and closed path
ETHOS was born from the incredible experience of OpenTx (and its main developer)
Business is business
for those like me who still believe in open source (and have made a very small contribution to OpenTx) an excellent alternative is EdgeTx 
Until business prevails over ethics again......
I just need to figure out how usable and manageable all the FRKY hardware I've purchased over time will remain..

phamiltonsmith, 2024-02-07:
> I have several of these receivers that I use with Opentx. What is the "current firmware" that you refer to?

Mike,
I have tried running OTx: FW and BL V 2.3.15, and Rx internal model ISRM-S-X9 1.1.0/2.1.10FCC
Rx is running version: 2024-01-09 	v1.0.10

Can you shed some light on your setup? Perhaps I am running something wrong!

BTW: I have also tried EdgeTx V 2.9.3

phamiltonsmith, 2024-02-07:
> Most likely a nightly firmware downloaded using a nightly version of Companion from https://github.com/opentx/opentx/releases/tag/nightly
> 
> It would be needed in order to have [29dd63a](https://github.com/opentx/opentx/commit/29dd63af8bf7217033e1b5a553db0ccb9454546a) and [6fef3b6](https://github.com/opentx/opentx/commit/6fef3b6849250bc3185553381b8a94f63da5fa4f), which allows for OTA firmware flashing and ACCESS configuration options for newer Archer receivers.
> 

Tried nightly version (https://github.com/opentx/opentx/releases/tag/nightly) but nada...

gagarinlg, 2024-02-07:
> ... an excellent alternative is EdgeTx Until business prevails over ethics again......

EdgeTX ha some structural differences. It is not controlled by one person, but by a committee of seven. That way a change of mnd of one person should not crush the whole project. We also have multiple people with admin rights to the EdgeTX GitHub repositories, the web site and some other accounts. We hope that this is stable enough to survive when someone is looking for new opportunities.

mpjf01, 2024-02-07:
Until recently I was under the impression that OpenTx was also managed by more than one person. Certainly the evidence of multiple developers and at least one other person with control abilities is in several places including here on Git.

Somewhere here I read however that Bertrand is the owner and as he has left the project nothing further can be done. It can't be that simple. More likely it is, in my view, that the various main developers have made a decision not to continue working on OpenTx software given the events that have followed from Bertrand's departure. This is likely a political decision but probably one based at least partly on the pragmatics associated with allowing development to continue while maintaining the sound control that we have seen over the last 12+ years over the project's principles and direction.

When Edge was first announced it was stated that it would be a vehicle for the development of new features in an environment which would eliminate the risks associated with major changes to a working system (OpenTx). All the new stuff could be tried out and used without messing up the software that most users were happy with. Then, after the new features (touch mainly but also others) had been used successfully in Edge, the changes would be incorporated into OpenTx. 

Maybe this is still the plan. Maybe something different has been stated subsequently that I've missed. Nevertheless we don't really know what the future really holds in this space. If anything is actually planned it's best not revealed until it's about to happen. Nobody wants to have to deal with the complaints that will explode when something doesn't happen exactly as documented or on time. 

I fly F3J and F5J sailplanes. These are among the most complex and resource consuming aircraft types as far as transmitter programming is concerned. OpenTx supports this application extremely well and while my transmitter hardware keeps functioning I see no reason to be concerned about the cessation of it's development. It may lack features that other people feel they need but this is really at the margin, any such lack is highly unlikely to have any real adverse impact on the vast majority of users.

For those that feel that they must have those lacking features, as stated above there are sound alternatives to OpenTx.

mpjf01, 2024-02-07:
> > I have several of these receivers that I use with Opentx. What is the "current firmware" that you refer to?
> 
> I have tried running OTx: FW and BL V 2.3.15, and Rx internal model ISRM-S-X9 1.1.0/2.1.10FCC Rx is running version: 2024-01-09 v1.0.10
> Can you shed some light on your setup? Perhaps I am running something wrong!
> 
> BTW: I have also tried EdgeTx V 2.9.3

Sorry. 

I have lost interest in FRSKY and don't follow their hardware developments closely any more. When I check I see that there is a new version of the GR6 receiver that was released last year. I have the older version of these receivers, so my comment above about having several GR6 receivers is not really relevant to your situation.

gagarinlg, 2024-02-07:
> When Edge was first announced it was stated that it would be a vehicle for the development of new features in an environment which would eliminate the risks associated with major changes to a working system (OpenTx). All the new stuff could be tried out and used without messing up the software that most users were happy with. Then, after the new features (touch mainly but also others) had been used successfully in Edge, the changes would be incorporated into OpenTx.\n\nMaybe this is still the plan. Maybe something different has been stated subsequently that I've missed. Nevertheless we don't really know what the future really holds in this space. If anything is actually planned it's best not revealed until it's about to happen. Nobody wants to have to deal with the complaints that will explode when something doesn't happen exactly as documented or on time.

EdgeTX is now too far apart from OpenTX to merge EdgeTX changes back into OpenTX, for the most part. There are no plans to even try this. 
The original idea was to have a co-development, where EdgeTX is a playground and OpenTX the stable alternative. At that point the progress on OpenTX 2.4 was already pretty close to non existent. Obviously the plan to port EdgeTX changes back did not work out, but EdgeTX grew to a good and strong open source project.
As far as we, the EdgeTX PSC, ars concerned, we are making our plans without consulting with OpenTX. We all know what Bertrand and all the other people that made OpenTX what it is now, did for the community and that this is the basis on which EdgeTX could become what it is now. But, as interests are shifting and time is limited, OpenTX seems to slowly go into a state of hibernation.

mpjf01, 2024-02-07:
> EdgeTX is now too far apart from OpenTX to merge EdgeTX changes back into OpenTX, for the most part. There are no plans to even try this. The original idea was to have a co-development, where EdgeTX [...]

Thank you for clearing that up, I hadn't read that this was the situation before. Maybe I should be paying more attention.

I did install Edge on one of my transmitters but found that there was just too much effort associated with converting all my models and checking that everything still functioned correctly. It might have been different if there was something in Edge that I wanted to use.

pfeerick, 2024-02-08:
I don't want this to seem like a OpenTX bashing thread... (especially given I'm also on the EdgeTX project steering committee, but have also been a long time OpenTX user)... so I would just like to add that OpenTX is still there as the "time proven stable" firmware for those who don't want the latest features. EdgeTX has come a long way in the three years it is underway, and there is no real sign of any slowdown in development - whether that be overhauling hard to maintain code in order to pave the way for new capabilities, migrating away from older, buggy libraries (and in doing so resolving some long standing issues), fixing some of the bugs that were never fixed in OpenTX, or adding new features based on community interest/demand or as people contribute to the project. 

However, there is nothing to stop someone forking OpenTX just like EdgeTX, and doing something  with that if they want to (or doing the same with EdgeTX)... such is the power of open source development... anyone can do anything they way ;)

MicheleVilla, 2024-02-08:
Powerful firmware like opentx or edgetx are intimately connected with the hardware on which they are installed
Mio therefore seems essential for radio manufacturers to be willing to provide maximum support to developers
Reverse engineering is always possible, but with current hardware developments it becomes more and more complicated.
And like frsky (which was born by cloning other brands...) for all producers the moment comes that they no longer like to be cloned.
it's true that OpenTx wasn't made up of just one developer, but it's clear that when he left the project effectively stopped.
Fortunately, thanks to the commitment of developers who continued with Edgetx, the project continues, but it moves away from FRSKY and this worries those like me who have invested a lot and find many FRSKY receivers, sensors and radios in the drawer.

kilrah, 2024-02-08:
OpenTX "permanent" team is/was more than one person, but unfortunately only one real professional software developer (Bertrand) with the skills to do major changes or review what the others did. There have been several very talented contributors who came and did one or 2 great things, but then once these things they were interested in were done were never seen again. So while there _are_ several people with project access who _could_ do things, there isn't anyone with both the skill and time to guide/review. Those who could have decided to create EdgeTX instead, I believe one reason being style/age. 
The whole 15 year lineage from the original th9x to er9x to gruvin9x to opentx were all "one guy's project with some others tagging along providing more or less small things", that's not how younger guys are taught/work anymore, and combining the 2 doesn't work well.

mpjf01, 2024-02-09:
Is started with a Turningy 9X transmitter and either Er9x or one of it's predecessors, can't recall exactly. I switched to OpenTx when it became available because it offered additional functionality that I wanted to use (flight modes). The changeover was stress free as I recall thanks to the developers making sure that the conversion was catered for. 

While I'm happy with OpenTx as it is I would make the not inconsiderable effort to switch to Edge if that offered functionality that I needed. The transition does not seem to be as effortless as in the previous cases.

The ability to keep using OpenTx as it is now is not just a end user decision though. There are facilities/servers that manage the firmware compilation and download services. Someone is managing and paying to operate these facilities. The funds required used to, I believe, come from donations from end users and some corporations including FRSKY. I have no idea who is managing these facilities right now or for how long they will continue to be available, but I ask here for whoever it is to give a reasonable warning of any plan to terminate them.

jiri-jirus, 2024-03-07:
it is unfortunate that FrSky is not able to keep up with Radiomaster and try to lock users into its obscure protocols.

geofrancis, 2024-03-07:
From what i can see there hasnt been a real update in over 6 months. so its dead as far as I am concerned.

gagarinlg, 2024-03-07:
> it is unfortunate that FrSky is not able to keep up with Radiomaster and try to lock users into its obscure protocols.

This has nothing to do with ability. They gained a lot of market share with open source based products and when other companies started to make radios compatible to that open source software, they protected their marketshare by moving a lot of the customers to a proprietary solution.

> From what i can see there hasnt been a real update in over 6 months. so its dead as far as I am concerned.

I think they might be an official patch release in the future with support for the newer FrSky receivers. I do not expect more releases to come afterwards.

phamiltonsmith, 2024-03-07:
My change to RadioMaster came sooner than I thought... I expect to receive my new gear next week... 

FRSky has one customer less...  I dont expect I'll be passing my Taranis on, it will probably go on my "Old radios" shelf! Along w/ my futaba Radios...

jiri-jirus, 2024-03-07:
> This has nothing to do with ability. They gained a lot of market share with open source based products and when other companies started to make radios compatible to that open source software, they [...]

Well I don't think they are able to protect their marketshare. everyone is using edgetx. frsky is dead.

LapinFou, 2024-05-31:
Like many people, I've been using OTX since version 1.52.
I'm planning to switch to EdgeTX; however, I've read that FrSky radios are not officially supported, which is understandable. My concern is whether EdgeTX is reliable on a FrSky radio (such as an older X9D+) for flying large models, or if it's risky. 🤔

3djc, 2024-05-31:
Pas de soucis avec ca mon Lapin ;)

LapinFou, 2024-06-02:
Merci JC ! Me voilà rassuré.👍

Point is taken. I will carefully review all my imported models.

skycraftmike, 2024-06-02:
Hi there 
what you guys sy to taking a STM32 chip same as whats in FlySky i6X radio and make a diy radio as arduino is getting slow compared to the STM32 chip could it be done saw you can program STM32 on the arduino IDE aswell

pfeerick, 2024-06-02:
No need to reinvent the wheel... while it is entirely possible, since you can program STM32 Blue Pill etc with Arduino IDE, it is also completely off-topic for this thread 🤪 There are plenty of projects online where people have done DIY radios, and [even used an ESP32](https://www.rcgroups.com/forums/showthread.php?4327269-EdgeTX-running-on-ESP32). 

https://howtomechatronics.com/projects/diy-arduino-rc-transmitter/
https://www.youtube.com/watch?v=sGTXAemf8Xk
https://www.reddit.com/r/esp32/comments/100kt61/esp32_diy_rc_transmitter_using_sx1280/
https://hackaday.com/2019/03/19/diy-six-channel-arduino-rc-transmitter/
https://github.com/cruwaller/elrs_handset