opentx/opentx #9037: Open TX Companion 2 - 3 - 15 opening page

unowat, 2024-03-17:
Opening page does not show the coloured icons. Previously they were OK.

pfeerick, 2024-03-22:
Check what Settings -> Set Icon Theme is set to if the icons are there, but look different. The default is "Yerico".

If the toolbar is completely missing, and you are on Windows, right click in the top menu area, and make sure there is a tick next to the File, Edit, Write and Help toolbars... 

[image]

unowat, 2024-03-22:
The Toolbar is completely missing and I use an Apple computer!

image.png (view on web)<https://github.com/opentx/opentx/assets/5500713/0e7960fe-7456-47b7-a006-2679a11bbd6f>