opentx/opentx #9040: Problem updating IRX4+ from OpenTX 2.3.15

Eingehirner, 2024-04-04:
Hello all

I am not sure which part actually is giving me problems. My issue is:

I have a Taranis X9D+ with FW OpenTX **2.3.15** and a IRangeX 4+ 4-in-1 external HF module with FW **1.3.1.78**. I tried to load the newest firmware 1.3.4.0 (file **mm-stm-ppm-aetr-v1.3.4.0.bin**) by copying it onto the Taranis SD card (FIRMWARE folder), using the Taranis to browse to the file, long Enter press, then select External Module. I get the error message that it's the wrong file and I would need the opentx-inv file. When I load the other available files (...serial...) they are not even shown on the Taranis. But the FW versions of Taranis and IRX should be compatible, in fact I have to be 2.3.9 or higher on the Taranis for the 1.3.4.0 to work on the IRX.

Since there is no file with this opentx-inv suffix to download (the Multi FW apparently changed away from this labelling before my already installed 1.3.1.78) I was thinking this must be an OpenTX issue. But somehow it doesn't fit together. There are people out there running the IRX4+ with 1.3.4.0 and somehow they got their module into this state and I assume a fair number of them used the Taranis for this as it's an attractive combination.

I would really appreciate any help in this matter!

Cheers
Simon

3djc, 2024-04-04:
Clearly ppm is a wrong file, otx expect serial ones

closed 2024-04-04 by 3djc

Eingehirner, 2024-04-04:
Yes, that worked! Unfortunately I now don't see any protocol beyond HoTT, which means that Traxxas is not there... but that's an IRX4 issue, not OpenTX ;)