opentx/opentx #9042: Bluetooth connection with BT Para Module not pairing with Smartphone (Taranis QX 7 Access)

Thomasa66, 2024-06-01:
Hello,

my goal was to use a internal BT module to get the telemetry data's to Telemetry Viewer App and for that reason my local dealer was asking FrSky which of the BT modules would do the job on my Taranis QX 7 Access. They adviced me to use the BT Para module. (In my case I need a internal module not the selfmade soulution HC-05/HC-06 via S.Port)

If I now choose Bluethooth Telemetry (tested with 100K Baud and 400K Baud) the BT name is "Hello" ...renaiming by seeting a other name doesent change anything, it is still broadcasting in BT network with the Name with "Hello".

But the main problem is: If I try pairing the BT module to my smartphone I receive instantly the message : "Failed to pair new device".

On the BT display on my Taranis I see the Local addr (think it is the MAC address of the BT module)
also I see Dist addr (which I think is the address of the smartphone after I have tried to pairing)
PIN code is shown as 000000

From the listed data's everything looks normal, but cant pairing /connecting.

OpenTX 2.3.15

Any ideas ?

Found some similiar posts ppl reporting problems in the internet but never with a solution.

Thomasa66, 2024-06-04:
Strange thing.. Meanwhile I figured out by random, I can connect my Transmitter QX 7 ACCESS -> Telemetry with the para Bluetooth modul to my WIndows 10 PC and even the name was changing from Hello to "myname".. (Which never happens trying to connect to Telemetry Viewer App)

I have no clue why connecting to WIndows 10 PC via Bluetooth for the Telemetry data's of the TX is good for, but connecting to Android -> Telemetry (which it should work) is not working ..

ChrisOhara57, 2024-09-13:
It does work with my Taranis X9D and x9Lite S (both Para module)

Keep in mind that the the radio has to be connected to the receiver first and telemetry has to be active (armed signal)
otherwise it will not establish a Bluetooth connection.

You have to start the App and select "connect" (do not try to connect in the Android settings menu)
Choose "Bluetooth LE"
select "Hello"