opentx/opentx #9044: Update Check Fails: Certificate Expired

jbeard6, 2024-06-26:
**Describe the bug**
The update check is broken because the SSL certificate has expired on [downloads.open-tx.org](https://downloads.open-tx.org). This is similar to #8723 and #9005, but the cause is that the SSL certificate has expired (at Tuesday, June 25, 2024 at 21:19:54 Eastern Daylight Time) and needs renewed.

**To Reproduce**
Steps to reproduce the behavior:
1. Open Companion
2. Click `Check for Updates...` menu item.
3. The application checks for updates to Companion.
4. The application presents an error: `Companion update check failed,new version information not found.` (sic)
5. Click OK.
6. The application checks for firmware updates.
7. The application presents an error: `Firmware update check failed,new version information not found or invalid.` (sic)

**Expected behavior**
Neither error should be presented. Rather, either update information or an up-to-date indication should be presented.

**General information**
 - OpenTX Companion 2.3.15 on macOS Sonoma 14.5

AveragePizza, 2024-06-26:
Ive got the same issue on Windows

jbeard6, 2024-06-26:
There's no update available (and likely won't be), so that seems unnecessary. But whoever manages that server should be aware that their cert is expired.

3djc, 2024-06-26:
He has been informed

AveragePizza, 2024-06-27:
> Temporary solution: change the current date in your operating system settings to June 24, 2024, then run Opentx Companion

This has worked for me thanks!

eimparas, 2024-09-08:
im getting the same . Is this project "dead" ?

gagarinlg, 2024-09-09:
Mostly.
The active fork of OpenTX is EdgeTX
edgetx.org