opentx/opentx #9046: A logical switch remains True  also when the AND condition becomes invalid. OpenTX 2.3.15, TX16S ver 1.

skjerr, 2024-07-15:
**Describe the bug**
A logical switch remains True  also when the AND condition becomes invalid.

**To Reproduce**
Logical switch programmed:
 L38   AND    SC up     SB --      SA down

Test sequence:
Start TX with all switches up.
Set SA down
Set SB --
L38 becomes True
Set SB up

**Expected behavior**
L38 becomes False,.
(but it remains True )

Additional info:
In Companion and in X9D+ it becomes False.
I made a walk-around by using 6P ¨100 instead of SA down
and that worked ok.

**Screenshots**
None.

**General information**
 - OpenTX version 2.3.15
 - Hardware TX16S version 1
 - OpenTX Companion 2.3.15   

A testPlane is attached. Look at LS 38.
[TestPlane.pdf](https://github.com/user-attachments/files/16234943/TestPlane.pdf)

mha1, 2024-07-16:
The last two changes to OpenTX happened 7 and 9 month ago. I wouldn't bet on a fix your your problem and even less on a new release. Your best option would be to upgrade to EdgeTX. btw: the logical switch setup you describe works as expected in EdgeTX

Bagira1230, 2024-10-23:
How about this walk-around with SA down?
L38 AND L39 SA down
L39 AND SC up SB --