opentx/opentx #9053: opentx

mike1-code, 2024-11-04:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version (e.g. 2.1.23 or N123)
 - Hardware (e.g. Frsky X9D+ or 9X Pro)
 - OpenTX Companion (e.g. 2.1.23 on Ubuntu 16.04)

mike1-code, 2024-11-04:
I have a problem after I flashed opendx on my horus x10s express, version OpenTX 2.3.15 appears (wrong PCB detected).

mawzthefinn, 2024-12-18:
Make sure you select the X10 Express in your radio profile, not the regular X10 + internalaccess