opentx/opentx #9059: Wrong FW links in the OpenTX Companion app versions 2.0-2.2

AzestUser, 2025-04-21:
#
I faced with an issue to download OpenTX v 2.1.9 for my old Turnigy 9X 

First of all you will see incorrect domain  if you try to check updates http://downloads-21.open-tx.org/firmware/stamp-opentx.txt 
Unable to check Updates

[image]

For requests monitoring I used Charles Proxy

[image]

[image]

Here is a feature of re-mapping and I used it to change  http://downloads-21.open-tx.org/firmware/stamp-opentx.txt  to the correct one https://downloads.open-tx.org/2.1/firmware/stamp-opentx.txt 

But the second failure is related for the directly Firmware hex file  
[image]
pretty the same issue, so I made assumption that it will fix an issue and remapped http://downloads-21.open-tx.org/firmware/getfw.php?fw=opentx-9x-en.hex  to http://downloads.open-tx.org/2.1/firmware/getfw.php?fw=opentx-9x-en.hex 
But unfortunately it doesn't. User can get only 301 server response  

[image]

<html>
<head><title>301 Moved Permanently</title></head>
<body bgcolor="white">
<center><h1>301 Moved Permanently</h1></center>
<hr><center>nginx/1.14.1</center>
</body>
</html>
 

Works correctly only for v 2.3 (but it doesn't support tugnigy with stock board)

Could you please advice a correct path for the hex file? I will replace it locally before the application fix. I see a lot of the same topics on the forums related to the Turnigy 9X in the internet, probably it would be useful not only for me.    

One more strange thing 
I can open either links and content is the same
https://downloads.open-tx.org/2.2/release/firmware_/stamp-opentx.txt 
https://downloads.open-tx.org/2.2/release/firmware/stamp-opentx.txt

https://downloads.open-tx.org/2.2/release/firmware -> 404 Not Found
But if you open https://downloads.open-tx.org/2.2/release/firmware_ you'll see
Index of /2.2/release/firmware_/
[../](https://downloads.open-tx.org/2.2/release/)
[lua_fields_x10.txt](https://downloads.open-tx.org/2.2/release/firmware_/lua_fields_x10.txt)                                 09-Jan-2018 09:48                7741
[lua_fields_x12s.txt](https://downloads.open-tx.org/2.2/release/firmware_/lua_fields_x12s.txt)                                09-Jan-2018 09:48                7741
[lua_fields_x7.txt](https://downloads.open-tx.org/2.2/release/firmware_/lua_fields_x7.txt)                                  09-Jan-2018 09:48                7329
[lua_fields_x9d.txt](https://downloads.open-tx.org/2.2/release/firmware_/lua_fields_x9d.txt)                                 09-Jan-2018 09:48                7513
[lua_fields_x9e.txt](https://downloads.open-tx.org/2.2/release/firmware_/lua_fields_x9e.txt)                                 09-Jan-2018 09:48                8016
[releasenotes.txt](https://downloads.open-tx.org/2.2/release/firmware_/releasenotes.txt)                                   12-Mar-2020 21:14               39458
[stamp-opentx.txt](https://downloads.open-tx.org/2.2/release/firmware_/stamp-opentx.txt)                                   21-May-2020 21:03                 202

kilrah, 2025-04-24:
> Deprecated, no more support will be given but firmware downloads are still available for the time being.

It was when that was written, but it's gone now.

AzestUser, 2025-04-24:
> > Deprecated, no more support will be given but firmware downloads are still available for the time being.
> 
> It was when that was written, but it eventually ended up breaking as expected. Might want to use er9x if you have a 9x.

Sorry, but where is release notes  on the site where was stated that old brunches are deprecated and not supported any more?

3djc, 2025-04-24:
Maybe here: https://www.open-tx.org/2018/02/26/avr-pull

AzestUser, 2025-04-24:
Maybe, but I understood this part - 'AVR radio users are advised to stay on 2.0 or 2.1 branch, for both Companion and firmware, as no new development for avr has been made since the last 2.0 versions'  as 2.1 is still available for downloading.

3djc, 2025-04-24:
That was 2018...

kilrah, 2025-04-24:
That was the point, "firmware downloads are still available for the time being" i.e. as long as it still worked, and it doesn't anymore. The server setup required to build firmware on demand for those platforms is gone.